OpenAMP/open-amp #346: rproc_virtio_get_features does not need to read dfeatures

tammyleino, 2022-02-04:
The rproc_virtio_get_features function reads both dfeatures and gfeatures from the vdev entry and returns bits set in both fields. It is not clear why this function reads dfeatures because gfeatures should already contain only those features which are set in dfeatures.

The description of these fields in remoteproc.h suggests that gfeatures should not provide more features than defined in dfeatures:

```
 * @dfeatures specifies the virtio device features supported by the firmware
 * @gfeatures is a place holder used by the host to write back the negotiated features that are supported by both sides.

```
It seems reading of dfeatures is not needed in the rproc_virtio_get_features function and can be removed.

edmooring, 2022-02-06:
This looks correct to me. @dfeatures is what the device supports, @gfeatures is what features the two ends have negotiated. Clients do not need to see features that might not be there.

tammyleino, 2022-02-07:
@edmooring then wouldn't just returning gfeatures be sufficient?  ANDing it with dfeatures doesn't return anything additional.

edmooring, 2022-02-08:
> @edmooring then wouldn't just returning gfeatures be sufficient? ANDing it with dfeatures doesn't return anything additional.

That's what I think as well. It looks to me like the set_features() call does this ANDing already. @arnopo what do you think?

arnopo, 2022-02-14:
This can be considered as a cross-check. Simplifying this means that we trust the remote side. I would be in favor of keeping it.

edmooring, 2022-02-20:
> This can be considered as a cross-check. Simplifying this means that we trust the remote side. I would be in favor of keeping it.

I see your point. That brings up an interesting architectural issue: How much do we or can we trust the remote side, and how do we protect ourselves against it?

arnopo, 2022-04-14:
I propose to close the issue an d keep current implementation

edmooring, 2022-04-14:
@tammyleino @arnopo I agree with closing this issue.

closed 2022-04-15 by arnopo