OpenAMP/open-amp #349: rpmsg_virtio_get_tx_payload_buffer assumes a full header will be present

tammyleino, 2022-02-09:
rpmsg_virtio_get_tx_payload_buffer assumes the payload will contain a full header and does not check whether the returned buffer size from rpmsg_virtio_get_tx_buffer is at least size of "rpmsg_hdr" structure before buffer size is decremented.  Is it impossible for a bad packet to be received at this layer?

tammyleino, 2022-02-17:
Is it necessary to use rpmsg_init_vdev_with_config though? Isn't it optional to use rpmsg_init_vdev instead?

arnopo, 2022-02-21:
you can use `rpmsg_init_vdev` which internally calls `rpmsg_init_vdev_with_config` with the `RPMSG_VIRTIO_DEFAULT_CONFIG` config

atulakella, 2026-10-02:
Checked it against current main (a0ec4fc), still present.

Device side, the unguarded subtraction:
https://github.com/OpenAMP/open-amp/blob/a0ec4fc2c764dd8927cafbd751637a29eb6bd761/lib/rpmsg/rpmsg_virtio.c#L410
`
*len -= sizeof(struct rpmsg_hdr);`

No check that *len is at least sizeof(struct rpmsg_hdr) before this.

It was suggested earlier that the host side gets a config time check and the device side gets an assert. Before I put together a fix; any preference on whether the device side should assert, or return an error instead?

arnopo, 2026-10-05:
> Checked it against current main ([a0ec4fc](https://github.com/OpenAMP/open-amp/commit/a0ec4fc2c764dd8927cafbd751637a29eb6bd761)), still present.

Good point https://github.com/OpenAMP/open-amp/pull/702 fix the RX part but not the TX part 

> Device side, the unguarded subtraction:
>
> https://github.com/OpenAMP/open-amp/blob/a0ec4fc2c764dd8927cafbd751637a29eb6bd761/lib/rpmsg/rpmsg_virtio.c#L410
>
>
> ` *len -= sizeof(struct rpmsg_hdr);`
> No [...]

it seems to me that we  should test the `len` returned by  `rpmsg_virtio_get_tx_buffer` and release the buffer  if  `len < sizeof(struct rpmsg_hdr); `

atulakella, 2026-10-05:
Agreed, a release and fail is better than assert or error. len is read from a vring descriptor in shared memory, so it's peer controlled input and shouldn't be an assert. Also, since rpmsg_virtio_get_tx_buffer() has already dequeued the buffer, returning an error alone would leak a descriptor.

Plan for the PR: in rpmsg_virtio_get_tx_payload_buffer(), after getting the buffer, if *len < sizeof(struct rpmsg_hdr), return the buffer to the ring via the existing release/return path for the current role, then return NULL. Otherwise the subtraction proceeds as before.

Will send the PR shortly.