OpenAMP/open-amp #350: Missing headers files in vitis 2021.1

jg-spitfire, 2022-02-12:
Hi, I am following the UG1186 v2021.1 page 44 from Xilinx which declares that for the app running in the remote processor this source file is required for the echo test together with:
https://github.com/OpenAMP/open-amp/blob/main/apps/machine/zynqmp_r5/platform_info.c
https://github.com/OpenAMP/open-amp/blob/main/apps/machine/zynqmp_r5/platform_info.h
https://github.com/OpenAMP/open-amp/blob/main/apps/machine/zynqmp_r5/rsc_table.c
https://github.com/OpenAMP/open-amp/blob/main/apps/machine/zynqmp_r5/rsc_table.h

but once I create an app in the r50 processor and use all those sources I have a lot of errors in the compilation, Vitis says there are missing .h files, so what sources do I need to reproduce this test?

jg-spitfire, 2022-02-12:
https://raw.githubusercontent.com/OpenAMP/open-amp/main/apps/examples/echo/rpmsg-echo.h is also included

arnopo, 2022-02-14:
Hi @jg-spitfire,
Do you contact Xilinx Support? your issue seems related to the implementation on a Xilinx Platform

edmooring, 2022-02-20:
The Xilinx fork of OpenAMP has diverged in a number of ways from upstream, and also lags upstream OpenAMP by a release or two. Mixing the library from Xilinx with the upstream apps may not always work.

edmooring, 2022-04-17:
We should close this issue.

closed 2022-04-19 by arnopo