OpenAMP/open-amp #352: What is the purpose of the buf argument in get_mem function of remoteproc_ops structure?

arnopo, 2022-02-15:
### Discussed in https://github.com/OpenAMP/open-amp/discussions/345

<div type='discussions-op-text'>

<sup>Originally posted by **tammyleino** February  4, 2022</sup>
The "get_mem" function in the remoteproc_ops structure has a "buf" argument. This argument is not documented in the input parameters. What is the purpose of "buf" argument in the "get_mem" function?</div>

arnopo, 2022-02-15:
Right `Buf` argument need to be documented.
This API is related to https://github.com/OpenAMP/open-amp/pull/169 proposed by @xiaoxiang781216 
Inspired by https://en.cppreference.com/w/c/chrono/gmtime

proposal:
```
	 * get_mem
	 *
	 * get remoteproc memory I/O region by either name, virtual
	 * address, physical address or device address.
	 *
	 * @rproc - pointer to remoteproc instance
	 * @name - memory name
	 * @pa - physical address
	 * @da - device address
	 * @va - virtual address
	 * @size - memory size
+	 * @buf - pointer to a local remoteproc_mem struct object to store the result
	 *
	 * @returns remoteproc memory pointed by buf if success, otherwise NULL
	 */

```

tammyleino, 2022-02-17:
@arnopo I guess I should do a PR first.

tammyleino, 2022-07-27:
@arnopo The PR has completed - this can be closed.

edmooring, 2022-08-06:
Closed per @tammyleino's request.

closed 2022-08-17 by arnopo