OpenAMP/open-amp #359: Verifying device type in resource table before creating rpmsg device

tammyleino, 2022-03-10:
remoteproc_create_virtio() gets the RSC_VDEV device at the indicated index from the resource table, but it does not check that it's actually an RPMSG device.  Can other types of devices be used with remoteproc?  If not, should remoteproc_create_virtio() ensure the device is actually of type VIRTIO_ID_RPMSG? Even if so, should rpmsg_init_vdev ensure the device type is VIRTIO_ID_RPMSG?

tammyleino, 2022-03-24:
@arnopo  Any thoughts on this?

arnopo, 2022-03-25:
The remoteproc_create_virtio should be able to support any virtio type, even If today only the RPMSG one seems used.

tammyleino, 2022-03-25:
Thank you @arnopo .  Is this for future compatibility or are other device types supported now?  When I look at virtio.h, I see the description of VIRTIO_ID_RPMSG as, "remote processor messaging", so it looked like this is the only device type supported with remoteproc.

arnopo, 2022-03-28:
`VIRTIO_ID_RPMSG` is related to the [virtio specification](https://docs.oasis-open.org/virtio/virtio/v1.1/csprd01/virtio-v1.1-csprd01.html#x1-1930005). 
Yes we expect in future to extend the virtio type supported by the library.

tammyleino, 2022-03-28:
Thank you. Understood.

closed 2022-03-28 by tammyleino