OpenAMP/open-amp #361: A method for sender to return a buffer acquired via rpmsg_get_tx_payload_buffer

tammyleino, 2022-03-18:
I don't see a way for a sender to return a buffer to the pool in the case of tx failure.  Is the only option to continue to try tx until the data is successfully sent?  Has rpmsg_release_tx_buffer API been considered?

arnopo, 2022-03-21:
Currently, nothing is implemented in Virtio to put the TX buffer back in the available list.Currently, nothing is implemented in virtio to put the TX buffer back in the available list.
One use case that could justify this API would be that the endpoint is unbound before the send.
So yes this is something that seems to me appropriate to add

arnopo, 2022-04-14:
FYI, I've started to investigate the problem and have a prototype that seems to address it.
unfortunately, not mature enough to send a pull request and too late for the 2022.04 release.
The corresponding code is available here: https://github.com/arnopo/open-amp/tree/tx_release
Any feedback on the solution is of course welcome.

closed 2022-10-18 by arnopo