OpenAMP/open-amp #379: Endpoint being removed without checking if the RPMSG_NS_DESTROY announcement was successfully sent

iqzaman, 2022-04-21:
The status of 'rpmsg_send_ns_message' is not being checked within _rpmsg_destroy_ept_. It seems that the function should not remove the endpoint (via the rpmsg_unregister_endpoint function call) in the case that 'rpmsg_send_ns_message' fails, i.e. the other side is not informed that the endpoint is being deleted.
With this implementation, when _rpmsg_destroy_ept_ is called, the endpoint will be removed, even if the other side is not informed about it.

tammyleino, 2022-04-22:
It looks to me that the error could be due to one of the following; no buffers available (recoverable), the Host disappearing, or an internal error that would probably cause a system failure (possibly caused by memory corruption). 

If the error is due to something recoverable like no buffer available, I think we should not destroy the endpoint, but return an error so the application can try again later.

Otherwise, we should check if the Host is still available, and if the Host is down, delete the endpoint.

Otherwise, return a severe error.  I think this case would likely require RPMsg to be shutdown and re-initialized.

edmooring, 2022-04-27:
I think it might be valuable to have the ability to unconditionally tear down and endpoint, for instance if we know the remote end has crashed. The linux API also returns void, and it would be nice to keep source compatibility.

Adding a new function, which can return an error, and does what @tammyleino describes above, seems to me to be the way to go. 

> Otherwise, we should check if the Host is still available, and if the Host is down, delete the endpoint.
This is trickier than it sounds. There is no mechanism in rpmsg, or rpmsg_virtio for this.

arnopo, 2023-05-09:
I close this issue as no answer. Please reopen it on need