OpenAMP/open-amp #390: what's plan to support the service/driver at the virtio level?

xiaoxiang781216, 2022-05-08:
In Linux kernel, we can:

1. Write various virtio drivers(e.g. virtio-blk, virtio-net...)
2. Write various rpmsg drivers(e.g. rpmsg-char, rpmsg-tty...)

But, we can just write rpmsg drivers in Open-AMP, but virtio drivers is very useful in some case:

1. Interact with virito device/driver in Linux kernel(A core) from RTOS(M core) in the same SoC
2. Improve IPC performance in some(e.g. network or file system) case

arnopo, 2022-05-10:
Hello @xiaoxiang781216 ,

Some other companies are also interested in extending the open-amp library to support more Virtio services.
Some discussions and works are just starting on the topic. 
I added people involved as assignee of the issue.

xiaoxiang781216, 2022-05-16:
Where can I find the related proposal or reference change? I want to evaluate the potential influence to the current code base. @arnopo @wmamills

danmilea, 2022-05-17:
There's ongoing work to integrate a virtio MMIO front-end layer in the open-amp library based on the Zephyr virtio MMIO implementation from [https://github.com/danmilea/zephyr/blob/main/drivers/virtio/virtio-mmio.c](url) (updated to use OpenAMP virtqueue APIs), with the goal to also include support for device drivers (e.g. virtio console, virtio-net).

The framework will also support hypervisorless mode operation. More information on hypervisorless virtio is available here: 
- [https://www.openampproject.org/news/hypervisorless-virtio-blog/](url)
- [https://github.com/danmilea/hypervisorless_virtio_zcu102](url)

@xiaoxiang781216

xiaoxiang781216, 2022-05-18:
Thanks @danmilea . I will study these document and feedback to you soon.

xiaoxiang781216, 2022-11-13:
@danmilea @arnopo I see that all virito driver support is done on virtio-exp branch, when will the work merge back to the main branch?

xiaoxiang781216, 2022-11-14:
@danmilea will you continue MMIO transport in the near feature?

danmilea, 2022-11-15:
@xiaoxiang781216 Do you mean adding support for the remoteproc style of virtio? Before looking into that, I think it's more relevant to investigate the requirements of an OpenAMP-based solution for VIRTIO back-ends, since the current implementation includes only front-end support. The scope pf my current work is to get everything on a usable level in a hypervisorless-virtio type of deployment.

CV-Bowen, 2023-03-07:
> There's ongoing work to integrate a virtio MMIO front-end layer in the open-amp library based on the Zephyr virtio MMIO implementation from [...]

@danmilea Hi, I want to implement the virtio-mmio-serial/net driver based on openamp. Do you have some examples show how to use the virtio mmio apis in virito-exp branch?

danmilea, 2023-03-09:
Hi @CV-Bowen 
[zephyr-openamp-staging](https://github.com/OpenAMP/zephyr-openamp-staging) (branch virtio-exp) includes [samples](https://github.com/OpenAMP/zephyr-openamp-staging/tree/virtio-exp/samples/virtio) for net and serial which use the open-amp library virtio mmio support in a standard virtual guest configuration. The hvl_net_rng_reloc sample works in a hypervisorless virtio environment (details in the sample [README.rst](https://github.com/OpenAMP/zephyr-openamp-staging/tree/virtio-exp/samples/virtio/hvl_net_rng_reloc#readme) file).

CV-Bowen, 2023-03-10:
@danmilea Got, Thanks a lot.

arnopo, 2023-06-22:
@CV-Bowen , @danmilea, @xiaoxiang781216, 

We have 2 complementary PRs for Virtio. I'd like to suggest that we move forward step by step to have a common dev approved  by everyone and ease the reviews by splitting PRs.

step 1:
- [ ] @CV-Bowen : create a new PR that introduce virtio API with create_virtqueues/delete_virtqueues dispatcher ops
      => no remorteproc_virtio or rpmsg update that will come in step 3

step 2:
- [ ] @danmilea : create a new PR with virtio MMIO transport only that use new dispatcher ops

step3
- [ ] @CV-Bowen  update #489( or create new one )   for the remoteproc virtio implementation and virtio RPMsg
- [ ] @danmilea: update #494 ( or create new one) for virtio services

arnopo, 2023-07-12:
Step 1 merged :  https://github.com/OpenAMP/open-amp/pull/495

closed 2023-10-24 by github-actions[bot]