OpenAMP/open-amp #426: Compilation issue with zephyr

mtreddy, 2022-10-05:
When building for zephyr I am seeing compilation issues as following.

CMake Error at lib/CMakeLists.txt:43 (add_dependencies)
  add_dependencies called with incorrect number of arguments 
-- at line number 42 --
add_dependencies(${ZEPHYR_CURRENT_LIBRARY}  ${OFFSETS_H_TARGET})
-- if I change this to --
add_dependencies(${ZEPHYR_CURRENT_LIBRARY} DEPENDS ${OFFSETS_H_TARGET})
It proceeds further

arnopo, 2022-10-06:
Hello,
Could you give more details to help to reproduce?
Zephyr version used ( or the sha1), your own updates, the build command used....

In addition could you send a Pull request with the fix you propose?

@carlocaione : do you detect this issue in your Zephyr environment?

Thanks,

mtreddy, 2022-10-06:
This time I have pulled zypherproject and SDK sephyr-sdk-0.15.0 . And I still see the issue. Could you pl let me know which version zephyr you use?
---

dini2@dini2:~/tmarri/src/open-amp$ cmake ./ -DWITH_ZEPHYR=on -DBOARD=qemu_cortex_m3 -DCMAKE_INCLUDE_PATH=~/tmarri/src/libmetal/build/lib/include/ -DCMAKE_LIBRARY_PATH=~/tmarri/src/libmetal/build/lib/ -Bbuild
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/dini2/tmarri/src/open-amp
-- Cache files will be written to: /home/dini2/.cache/zephyr
-- Zephyr version: 3.2.99 (/home/dini2/tmarri/src/zephyrproject/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: qemu_cortex_m3
-- Found host-tools: zephyr 0.15.0 (/home/dini2/tmarri/src/zephyr-sdk-0.15.0)
-- Found toolchain: zephyr 0.15.0 (/home/dini2/tmarri/src/zephyr-sdk-0.15.0)
-- Found BOARD.dts: /home/dini2/tmarri/src/zephyrproject/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3.dts
-- Generated zephyr.dts: /home/dini2/tmarri/src/open-amp/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/dini2/tmarri/src/open-amp/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/dini2/tmarri/src/open-amp/build/zephyr/dts.cmake
Parsing /home/dini2/tmarri/src/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/dini2/tmarri/src/open-amp/build/zephyr/.config'
No change to configuration in '/home/dini2/tmarri/src/open-amp/build/zephyr/.config'
No change to Kconfig header in '/home/dini2/tmarri/src/open-amp/build/zephyr/include/generated/autoconf.h'
-- open-amp version: 1.2.0 (/home/dini2/tmarri/src/open-amp)
-- Host:    Linux/x86_64
-- Target:  Generic/arm
-- Machine: cortexm
-- C_FLAGS :  -Wall -Wextra
-- Found Libmetal: /home/dini2/tmarri/src/libmetal/build/lib/libmetal.a  
-- Looking for include file stdatomic.h
-- Looking for include file stdatomic.h - found
-- Looking for include file fcntl.h
-- Looking for include file fcntl.h - found
CMake Error at lib/CMakeLists.txt:42 (add_dependencies):
  add_dependencies called with incorrect number of arguments

-- Configuring incomplete, errors occurred!
See also "/home/dini2/tmarri/src/open-amp/build/CMakeFiles/CMakeOutput.log".
See also "/home/dini2/tmarri/src/open-amp/build/CMakeFiles/CMakeError.log".
dini2@dini2:~/tmarri/src/open-amp$ vi lib/CMakeLists.txt
dini2@dini2:~/tmarri/src/open-amp$ cmake ./ -DWITH_ZEPHYR=on -DBOARD=qemu_cortex_m3 -DCMAKE_INCLUDE_PATH=~/tmarri/src/libmetal/build/lib/include/ -DCMAKE_LIBRARY_PATH=~/tmarri/src/libmetal/build/lib/ -Bbuild
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/dini2/tmarri/src/open-amp
-- Cache files will be written to: /home/dini2/.cache/zephyr
-- Zephyr version: 3.2.99 (/home/dini2/tmarri/src/zephyrproject/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: qemu_cortex_m3
-- Found host-tools: zephyr 0.15.0 (/home/dini2/tmarri/src/zephyr-sdk-0.15.0)
-- Found toolchain: zephyr 0.15.0 (/home/dini2/tmarri/src/zephyr-sdk-0.15.0)
-- Found BOARD.dts: /home/dini2/tmarri/src/zephyrproject/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3.dts
-- Generated zephyr.dts: /home/dini2/tmarri/src/open-amp/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/dini2/tmarri/src/open-amp/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/dini2/tmarri/src/open-amp/build/zephyr/dts.cmake
Parsing /home/dini2/tmarri/src/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/dini2/tmarri/src/open-amp/build/zephyr/.config'
No change to configuration in '/home/dini2/tmarri/src/open-amp/build/zephyr/.config'
No change to Kconfig header in '/home/dini2/tmarri/src/open-amp/build/zephyr/include/generated/autoconf.h'
-- open-amp version: 1.2.0 (/home/dini2/tmarri/src/open-amp)
-- Host:    Linux/x86_64
-- Target:  Generic/arm
-- Machine: cortexm
-- C_FLAGS :  -Wall -Wextra
-- Configuring done
CMake Error at /home/dini2/tmarri/src/zephyrproject/zephyr/cmake/modules/extensions.cmake:418 (add_library):
  No SOURCES given to target: app
Call Stack (most recent call first):
  /home/dini2/tmarri/src/zephyrproject/zephyr/cmake/modules/kernel.cmake:209 (zephyr_library_named)
  /home/dini2/tmarri/src/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:117 (include)
  /home/dini2/tmarri/src/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/dini2/tmarri/src/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  /home/dini2/tmarri/src/zephyrproject/zephyr/cmake/app/boilerplate.cmake:25 (find_package)
  cmake/syscheck.cmake:8 (include)
  CMakeLists.txt:18 (include)

CMake Generate step failed.  Build files cannot be regenerated correctly.

mtreddy, 2022-10-07:
When I printed the messages for ${ZEPHYR_CURRENT_LIBRARY} ${OFFSETS_H_TARGET}and in lib/CMakeLists.txt

if (WITH_ZEPHYR)
  zephyr_library_named(${OPENAMP_LIB})
  message(STATUS "openamp_li :${OPENAMP_LIB}")
  message(STATUS "zephyr_cur: ${ZEPHYR_CURRENT_LIBRARY} ")
  message(STATUS "offset_h: ${OFFSETS_H_TARGET}")
  add_dependencies(${ZEPHYR_CURRENT_LIBRARY} ${OFFSETS_H_TARGET})
  zephyr_library_sources(${_sources})
  zephyr_include_directories(${CMAKE_CURRENT_SOURCE_DIR}/include)

#### in the output there is no value assigned to  OFFSETS_H_TARGET ###
-- openamp_li :open_amp
-- zephyr_cur: open_amp
-- offset_h:
CMake Error at lib/CMakeLists.txt:45 (add_dependencies):
  add_dependencies called with incorrect number of arguments

--

that could be why config is failing.

carlocaione, 2022-10-07:
Zephyr is already shipping its own libmetal and openamp shims (see https://github.com/zephyrproject-rtos/libmetal and https://github.com/zephyrproject-rtos/open-amp) so you should not pull anything from the official repos since both are already correctly integrated in zephyr.

Please see the samples in https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/ipc for the correct way to integrate openamp with Zephyr.

mtreddy, 2022-10-10:
Thanks for the help.  I am able to build in Zephyr tree it self  using " west build -b mps2_an521 samples/subsys/ipc/rpmsg_service" .  BTW there is typo in README it say ZEPHRY it should be ZEPHYR.

arnopo, 2022-10-13:
Hi,
Thanks @carlocaione for your help. As you mentioned and as it partially explained at the beginning of the README Zephyr section in the README.md, the zephyr environment as to be used in this case
I will clean the second part that involves too much cross-dependency between OpenAMP and Zephyr

closed 2022-10-18 by arnopo