OpenAMP/open-amp #427: remoteproc_load issue not applied to remoteproc_load_noblock

tammyleino, 2022-10-07:
The PR https://github.com/OpenAMP/open-amp/pull/165/ adds code to store the metal i/o region of the resource table in the remoteproc structure to the function remoteproc_load, but this same fix was not added to the function remoteproc_load_noblock.  Was this intentional?

arnopo, 2022-10-13:
Good catch!  Yes, the fix seems to be missing. 

@edmooring, @TanmayShah-xilinx do you know if AMD uses this interface?

TanmayShah-xilinx, 2022-10-13:
@tammyleino Thanks for catching this.

@arnopo This Interface is used in file: apps/examples/load_fw/load_fw.c func: load_exectuable_noblock

However, I see that it is used only with flag: #ifndef RPU_BOOT_LINUX So, I believe it is not used for ZynqMP platform. However, other platforms might be affected.

edmooring, 2022-10-16:
The RPU_BOOT_LINUX flag is for the case when the R5 boots linux rather than the usual approach of Linux starting the R5 via remoteproc. If I remember correctly, this example is for booting the remote processor using the OpenAMP library implementation of remoteproc, rather than the Linux kernel implementation. 

We should probably fix this, but it is not urgent, since this is a little-used part of the library.

closed 2023-02-27 by arnopo