OpenAMP/open-amp #463: Linker error `undefined reference to '__atomic_store_8'`

sergeyblagodarnyy, 2023-02-02:
Hello, dear colleagues.

I'm trying to link `openamp/rpmsg` libset into my program for STM32-MP1 M4 micro controller and stuck with this kind of error. I've spend  three days for solving with no results. I tried differend versions of the libset: 0.1.0; 1.1.0; 1.3.0 - always the same result.

What I do:
1) Prepare toolchain file for libmetal
2) Compile libmetal into a static library following it's instruction. https://github.com/OpenAMP/libmetal#readme
3) Prepare toolchain file for open-amp
4) Compile open-amp into a static library following it's instruction. https://github.com/OpenAMP/open-amp#readme
5) Prepare a minimal "hello_world" example to demonstrate this issue. Prepare CMakeLists.txt and toolchain file for my project. Link my project with the mentioned libraries.

```
#include "stm32mp1xx_hal.h"
#include "virt_uart.h"
#include "ipcc.h"

int main()
{
   HAL_Init();

   assert(!IS_ENGINEERING_BOOT_MODE());

   __HAL_RCC_HSEM_CLK_ENABLE();
   MX_IPCC_Init();
   MX_OPENAMP_Init(RPMSG_REMOTE, NULL);

   while(1)
   {
   }
}
```
And again in compiling I have this.
```
/home/sergey/projects/developUTC/source/arm-tools/gcc/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /home/sergey/projects/developUTC/source/arm-tools/STM32CubeMP1/Middlewares/Third_Party/OpenAMP/open-amp/install/m4-generic-release/usr/local/lib/libopen_amp.a(remoteproc_virtio.c.obj): in function `metal_io_read':
remoteproc_virtio.c:(.text+0x162): undefined reference to `__atomic_load_8'
/home/sergey/projects/developUTC/source/arm-tools/gcc/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /home/sergey/projects/developUTC/source/arm-tools/STM32CubeMP1/Middlewares/Third_Party/OpenAMP/open-amp/install/m4-generic-release/usr/local/lib/libopen_amp.a(remoteproc_virtio.c.obj): in function `metal_io_write':
remoteproc_virtio.c:(.text+0x28e): undefined reference to `__atomic_store_8'
```

Do you know where these symbols must be? I tried to find them in toolchain binaries, but haven't got any comprehensible results. They are absent in arm-none-eabi-gcc and in its libraries.

What to do with this problem?

I attach toolchain files for libmetal, open-amp and hello_world projects. Also I attach the build console outputs.
libmetal:
[utc-m4.cmake.txt](https://github.com/OpenAMP/open-amp/files/10566670/utc-m4.cmake.txt)
[make.log](https://github.com/OpenAMP/open-amp/files/10566645/make.log)
[cmake.log](https://github.com/OpenAMP/open-amp/files/10566646/cmake.log)
open-amp:
[utc-m4.cmake.txt](https://github.com/OpenAMP/open-amp/files/10566691/utc-m4.cmake.txt)
[make.log](https://github.com/OpenAMP/open-amp/files/10566693/make.log)
[cmake.log](https://github.com/OpenAMP/open-amp/files/10566694/cmake.log)
hello-world:
[CMakeLists.txt](https://github.com/OpenAMP/open-amp/files/10566699/CMakeLists.txt)
[utc-m4.cmake.txt](https://github.com/OpenAMP/open-amp/files/10566701/utc-m4.cmake.txt)
[make.log](https://github.com/OpenAMP/open-amp/files/10566703/make.log)
[cmake.log](https://github.com/OpenAMP/open-amp/files/10566704/cmake.log)

Toolchain: gcc-arm-none-eabi-10-2020-q4-major

Hope to hear from you soon!

arnopo, 2023-02-02:
If I well understood:
- build and link Libmetal static lib : OK
- build and link open-amp static lib: OK
- build your application: OK
-link the libraries and your application: KO

Seems to me as a first analysis that you need to add the "atomic" library 

Adding following line in your `hello-world/CmakeLists.txt` could help

`set(CMAKE_CXX_LINK_FLAGS "${CMAKE_CXX_LINK_FLAGS} -latomic")`

If it still does not work , could you share your project on your Github?

sergeyblagodarnyy, 2023-02-02:
Thank you for reply @arnopo 

No, with this option it couldn't find the libatomic library.
`set(CMAKE_CXX_LINK_FLAGS "${CMAKE_CXX_LINK_FLAGS} -latomic")`

output:
```
FAILED: mt_hello_world.elf 
: && /home/sergey/projects/developUTC/source/arm-tools/gcc/bin/arm-none-eabi-g++ --sysroot=/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/ -g  -latomic -flto    --sysroot=\"/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi\" -mthumb -mabi=aapcs -Wl,--gc-sections -Og -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -T /home/sergey/projects/developUTC/source/arm-tools/stm32-cmake/cmake/stm32/stm32mp15xx_m4.ld --specs=nosys.specs CMakeFiles/mt_hello_world.dir/error_handler.c.obj CMakeFiles/mt_hello_world.dir/hello_world.c.obj CMakeFiles/mt_hello_world.dir/stm32mp1xx_it.c.obj CMakeFiles/mt_hello_world.dir/openamp/rpmsg.cpp.obj CMakeFiles/mt_hello_world.dir/openamp/printf.cpp.obj CMakeFiles/mt_hello_world.dir/openamp/stm32mp1xx_hal_msp.c.obj CMakeFiles/mt_hello_world.dir/home/sergey/projects/developUTC/source/arm-tools/STM32CubeMP1/Middlewares/Third_Party/OpenAMP/virtual_driver/virt_uart.c.obj CMakeFiles/mt_hello_world.dir/openamp/openamp.c.obj CMakeFiles/mt_hello_world.dir/openamp/ipcc.c.obj CMakeFiles/mt_hello_world.dir/openamp/mbox_ipcc.c.obj CMakeFiles/mt_hello_world.dir/openamp/rsc_table.c.obj CMakeFiles/mt_hello_world.dir/openamp/rpmsg_hdr.c.obj CMakeFiles/mt_hello_world.dir/home/sergey/projects/developUTC/source/arm-tools/STM32CubeMP1/Drivers/STM32MP1xx_HAL_Driver/Src/stm32mp1xx_hal.c.obj CMakeFiles/mt_hello_world.dir/home/sergey/projects/developUTC/source/arm-tools/STM32CubeMP1/Drivers/CMSIS/Device/ST/STM32MP1xx/Source/Templates/system_stm32mp1xx.c.obj CMakeFiles/mt_hello_world.dir/home/sergey/projects/developUTC/source/arm-tools/STM32CubeMP1/Drivers/STM32MP1xx_HAL_Driver/Src/stm32mp1xx_hal_ipcc.c.obj CMakeFiles/mt_hello_world.dir/home/sergey/projects/developUTC/source/arm-tools/STM32CubeMP1/Drivers/STM32MP1xx_HAL_Driver/Src/stm32mp1xx_hal_cortex.c.obj CMakeFiles/mt_hello_world.dir/home/sergey/projects/developUTC/source/arm-tools/STM32CubeMP1/Drivers/STM32MP1xx_HAL_Driver/Src/stm32mp1xx_hal_rcc.c.obj -o mt_hello_world.elf  libmcu_hal.a  /home/sergey/projects/developUTC/source/arm-tools/STM32CubeMP1/Middlewares/Third_Party/OpenAMP/open-amp/install/m4-generic-release/usr/local/lib/libopen_amp.a  /home/sergey/projects/developUTC/source/arm-tools/STM32CubeMP1/Middlewares/Third_Party/OpenAMP/libmetal/install/m4-generic-release/usr/local/lib/libmetal.a  libmcu_hal.a  /home/sergey/projects/developUTC/source/arm-tools/STM32CubeMP1/Drivers/CMSIS/Device/ST/STM32MP1xx/Source/Templates/gcc/startup_stm32mp157axx_cm4.s && :
/home/sergey/projects/developUTC/source/arm-tools/gcc/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: cannot find -latomic
```

And indeed it is absent it the toolchain. Should it exist?
```
sergey@sergey-HP:~/projects/developUTC/sandbox/hello_world/build$ find ~/projects/developUTC/source/arm-tools/gcc -name *atomic*
/home/sergey/projects/developUTC/source/arm-tools/gcc/share/doc/gcc-arm-none-eabi/html/gcc/_005f_005fatomic-Builtins.html
/home/sergey/projects/developUTC/source/arm-tools/gcc/lib/gcc/arm-none-eabi/10.2.1/include/stdatomic.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/atomic
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/ext/atomicity.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/bits/atomic_lockfree_defines.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/bits/atomic_base.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/bits/shared_ptr_atomic.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/bits/atomic_futex.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v7-a+fp/hard/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v7-a+fp/softfp/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v7e-m+fp/hard/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v7e-m+fp/softfp/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v7/nofp/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v7ve+simd/hard/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v7ve+simd/softfp/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v7e-m+dp/hard/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v7e-m+dp/softfp/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v7-a/nofp/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v6-m/nofp/bits/atomic_word.h
[... cut ...]
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/thumb/v7-a+simd/softfp/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/arm/v5te/hard/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/arm/v5te/softfp/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/c++/10.2.1/arm-none-eabi/bits/atomic_word.h
/home/sergey/projects/developUTC/source/arm-tools/gcc/arm-none-eabi/include/stdatomic.h
```

sergeyblagodarnyy, 2023-02-02:
>     * build and link Libmetal static lib : OK
>     * build and link open-amp static lib: OK
>     * build your application: OK
>       -link the libraries and your application: KO
Exactly!

> If it still does not work , could you share your project on your Github?
Yes, I'm preparing it.

I found a thread where also linker couldn't find the reference to some atomic functions. It seems arm-none-eabi-gcc doesn't have these functions for some arm cores. Especially if that cores don't support such atomic instructions...
https://community.st.com/s/question/0D53W00001ExC9XSAV/how-to-fix-linker-error-with-atomic-compareexchangeweak

arnopo, 2023-02-02:
> Yes, I'm preparing it.

Thanks !

> 
> I found a thread where also linker couldn't find the reference to some atomic functions. I don't know is it related to my case or not, but it seems arm-none-eabi-gcc doesn't have some atomic functions for some arm cores. Especially if that cores don't support such atomic instructions on hardware level... https://community.st.com/s/question/0D53W00001ExC9XSAV/how-to-fix-linker-error-with-atomic-compareexchangeweak

The stm32cube IDE uses gcc to  generate project for the cortex M4, without atomic issue.

Could be related to the use of g++   ....
Regarding your log files  I can see that you build the libraries why gcc and the application with g++.
- Did you try to build your application with gcc?
- Did you try to build the libraires with g++?

The issue canbe around this file:
https://github.com/OpenAMP/libmetal/blob/main/lib/atomic.h
There can be a mismatch between the include and the lib needed by the linker if you build with gcc and link with g++.

sergeyblagodarnyy, 2023-02-06:
> Thanks !
>
>
> The stm32cube IDE uses gcc to generate project for the cortex M4, without atomic issue.
>
> Could be related to the use of g++ .... Regarding your log files I can see that you build the [...]

Hello, dear  Arnaud
I've prepared a minimal example where you can see the error. I removed all C++ code so now it doesn't use g++ utility. Unfortunately same result.
Please, make the following steps:
1) Download the example
```
$ git clone git@github.com:Faramax/open_amp_build_test.git
$ cd open_amp_build_test/
/open_amp_build_test$ git submodule init
/open_amp_build_test$ git submodule update
```
2) Set in the toolchain files following variables:
```
/open_amp_build_test/libmetal/cmake/platforms/utc-m4.cmake
/open_amp_build_test/open-amp/cmake/platforms/utc-m4.cmake
/open_amp_build_test/utc-m4.cmake
```

`CROSS_PREFIX` - your toolchain prefix. For example: "/home/sergey/projects/developUTC/source/arm-tools/gcc/bin/arm-none-eabi-"
`STM32_CUBE_MP1_PATH` - path to your stm32mp1 cube library.

3) Build
```
/open_amp_build_test$ cd libmetal/
/open_amp_build_test/libmetal$ mkdir build 
/open_amp_build_test/libmetal$ cd build/
/open_amp_build_test/libmetal/build$ cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/platforms/utc-m4.cmake ../
/open_amp_build_test/libmetal/build$ make VERBOSE=1 DESTDIR=../install install
/open_amp_build_test/libmetal/build$ cd ../../open-amp/
/open_amp_build_test/open-amp$ mkdir build
/open_amp_build_test/open-amp$ cd build/
/open_amp_build_test/open-amp/build$ cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/platforms/utc-m4.cmake ../
/open_amp_build_test/open-amp/build$ make VERBOSE=1 DESTDIR=../install install
/open_amp_build_test/open-amp/build$ cd ../../
/open_amp_build_test$ mkdir build
/open_amp_build_test$ cd build/
/open_amp_build_test/build$ cmake -DCMAKE_TOOLCHAIN_FILE=../utc-m4.cmake ../
/open_amp_build_test/build$ make VERBOSE=1
```

mj-osborne, 2023-02-06:
Hello, 

I'm also having a similar issue... Using the latest OpenAMP (2022.10.1 1.3.1)/Libmetal (2022.10.0 1.3.0) on a Cortex-M4.  
I have verified that https://github.com/OpenAMP/libmetal/pull/127 is included.

Compiling Libmetal/OpenAMP as C static libs with gcc, application is C++.

I was initially using gcc 12.2.rel1 and had the same atomic issue when linking against Libmetal (no OpenAMP yet):
`undefined reference to '__atomic_load_8'`
`undefined reference to '__atomic_store_8'`

I downgraded to gcc 11.3.rel1 and it appeared the issue was resolved; however, after a bit of poking around I can reproduce it when `-O0` is specified.

I then added OpenAMP to the project and have additional linker errors:
`lib/libopen_amp.a(rpmsg_virtio.c.obj): in function 'rpmsg_virtio_send_offchannel_nocopy': .../rpmsg_virtio.c:426: undefined reference to 'metal_io_block_write'`
`lib/libopen_amp.a(rpmsg_virtio.c.obj): in function 'rpmsg_virtio_ns_callback': .../rpmsg/rpmsg_virtio.c:632: undefined reference to 'metal_io_block_read'`
`lib/libopen_amp.a(rpmsg_virtio.c.obj): in function 'rpmsg_virtio_send_offchannel_raw': .../rpmsg/rpmsg_virtio.c:513: undefined reference to 'metal_io_block_write'`
`lib/libopen_amp.a(rpmsg_virtio.c.obj): in function 'rpmsg_init_vdev_with_config': ... /rpmsg/rpmsg_virtio.c:822: undefined reference to 'metal_io_block_set'`
`lib/libopen_amp.a(virtio.c.obj): in function 'virtio_create_virtqueues': .../virtio/virtio.c:111: undefined reference to 'metal_io_block_set'`

Discussing with a co-worker and we're thinking we need to add our own implementations of `__atomic_load_8` & `__atomic_store_8`.

arnopo, 2023-02-07:
Hello
Thanks @sergeyblagodarnyy for you environment
For people who want to reproduce, you need also to clone stm32Cubemp1 
`git clone git@github.com:STMicroelectronics/STM32CubeMP1.git`
and update `STM32_CUBE_MP1_PATH`  in consequence.

I reproduce the issue. It seems linked to the gcc that provides the `stdatomic.h` inclide file but not the associated library.

One solution is to include the https://github.com/OpenAMP/libmetal/blob/main/lib/compiler/gcc/atomic.h
This is done by undefining the `HAVE_STDATOMIC_H`, which is set by the cmake  in https://github.com/OpenAMP/libmetal/blob/main/cmake/depends.cmake

There is another solution which explains behavior observed by @mj-osborne ( but I discourage to use it). 
Adding the compilation optimization -O3 in `CMAKE_C_FLAGS` for the open-amp library build also solves the issue.
The reason is more obscure. I suppose that the open-amp lib is optimized and `__atomic_load_8 ` & `__atomic_store_8  `are  "inlined".  
This can also explain why the issue is not reproduced using the stm32Cube IDE. The OpenAMP libraries are not generated but the source code is directly integrated in projects.

In both case I tested only build. So hope that it will solve your issue.

arnopo, 2023-02-24:
Great if you are now unlocked, Can we close the issue?

closed 2023-02-24 by sergeyblagodarnyy