OpenAMP/open-amp #486: doxygen should use relevant "module" defs to organize the docs

wmamills, 2023-04-25:
Zephyr uses doxygengroup to integrate doxygen content in place in the sphinx docs.
We plan to do this as well.

libmetal already has well defined "groups" aka "modules" that organize the content into API sub groups.  

open-amp  should do the same.  This can be done and tested before Sphinx integration by just browsing the "modules" pull down menu in the default doxygen content.

wmamills, 2023-10-24:
This is not done

wmamills, 2023-10-24:
Please compare: https://openamp.readthedocs.io/en/latest/doxygen/openamp/modules.html
And https://openamp.readthedocs.io/en/latest/doxygen/libmetal/modules.html

libmetal is much better organized and easier to understand.

arnopo, 2026-10-23:
@wmamills : with the rework of the documentation I think we can close this one.