OpenAMP/open-amp #487: Question about the resource table

shellyfengfeng, 2023-05-09:
Resource table seems defined by remote, why not the master define the related resouce such as shared memory and vring address?
Thanks.

shellyfengfeng, 2023-05-09:
Thanks for your reply. But It seems the section ".resource_table" is used in the remote to tell the master the vring and share memory information. 发自我的手机-------- 原始邮件 --------发件人： Tanmay ***@***.***>日期： 2023年5月9日周二 晚上10:33收件人： OpenAMP/open-amp ***@***.***>抄送： shellyfengfeng ***@***.***>, Author ***@***.***>主    题： Re: [OpenAMP/open-amp] Question about the resource table (Issue #487)
I am assuming master is linux kernel. In that case, remoteproc node in device-tree contains "memory-region" property which has vring addresses.

arnopo, 2023-05-09:
@shellyfengfeng 
Please could you resent a readable comment

shellyfengfeng, 2023-05-10:
I mean master will load the remote image and get resource table from the remote image.
If the master has such definition, the master doesn't need to load the resource table from the remote image.

edmooring, 2023-05-10:
The remote side contains configuration information about things like how many virtio devices are expected that the host side has no way of obtaining other than by reading the resource table. The Linux kernel can and does also write to the resource table to inform the remote side of such things as shared memory addresses.

shellyfengfeng, 2023-05-10:
> The remote side contains configuration information about things like how many virtio devices are expected that the host side has no way of obtaining other than by reading the resource table. The [...]

Assuming the host and the remote are baremental, after the host load the remote image, it will load the ".resource_table" seciton and  use the "rsc_table" to initial the vring and virtqueues.

arnopo, 2023-07-12:
@shellyfengfeng 
Are you waiting for more information or can we close the issue?

shellyfengfeng, 2023-07-13:
Yes, please close the issue.

closed 2023-07-13 by shellyfengfeng