OpenAMP/open-amp #499: Need to do a dcache invalidation when reading status from resource table

iuliana-prodan, 2023-07-12:
For HiFi4 DSP, the resource table is added in SRAM which is cacheable region.
When reading the status I need to do a dcache invalidation otherwise the status is never updated.

The issue was found while enabling the [openamp_rsc_table](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/ipc/openamp_rsc_table) from Zephyr.
This app is running on HiFi4 DSP and on Cortex A we have Linux with `rpmsg_client_sample`.

iuliana-prodan, 2023-07-12:
Here's my proposed fix: https://github.com/OpenAMP/open-amp/pull/500

iuliana-prodan, 2023-07-12:
> Here's my proposed fix: #500

I would like to do this conditional, something like:

```
diff --git a/lib/remoteproc/remoteproc_virtio.c b/lib/remoteproc/remoteproc_virtio.c
index 9dd0f36..4730116 100644
--- a/lib/remoteproc/remoteproc_virtio.c
+++ b/lib/remoteproc/remoteproc_virtio.c
@@ -41,7 +41,11 @@ static unsigned char rproc_virtio_get_status(struct virtio_device *vdev)
        rpvdev = metal_container_of(vdev, struct remoteproc_virtio, vdev);
        vdev_rsc = rpvdev->vdev_rsc;
        io = rpvdev->vdev_rsc_io;
-       metal_cache_invalidate(vdev_rsc, sizeof(vdev_rsc));
+
+#ifdef VIRTIO_CACHED_RSC_TABLE
+       metal_cache_invalidate(vdev_rsc, sizeof(vdev_rsc));
+#endif /* VIRTIO_CACHED_RSC_TABLE */
+
        status = metal_io_read8(io,
                                metal_io_virt_to_offset(io, &vdev_rsc->status));
        return status;
```

But, how can I activate the OpenAMP macro (the VIRTIO_CACHED_RSC_TABLE ) from my platform or sample in Zephyr?

carlocaione, 2023-07-12:
I'll leave to @arnopo the open-amp part of the question.

For this part:

> But, how can I activate the OpenAMP macro (the VIRTIO_CACHED_RSC_TABLE ) from my platform or sample in Zephyr?

If you have a `VIRTIO_CACHED_RSC_TABLE` define in open-amp you have to add support for it in the `options.cmake` file in open-amp, for example:
https://github.com/OpenAMP/open-amp/blob/7f906105a9ffef8e59287f0fca68fc8f8df7725c/cmake/options.cmake#L83-L91

and add support for that in Zephyr tweaking the `CMakeLists.txt` file for the open-amp module, for example:
https://github.com/zephyrproject-rtos/open-amp/blob/c904a01d4a882bcbb39987e0e2ce5308f49ac7ad/CMakeLists.txt#L20-L23

iuliana-prodan, 2023-07-18:
Fixed by #500