OpenAMP/open-amp #501: Move examples to openamp-system-reference repo.

TanmayShah-xilinx, 2023-07-14:
Openamp library isn't the right place to have demo and examples.

Move examples to openamp-system-reference repository without breaking backward compatibility.

tnmysh, 2024-06-19:
@arnopo , @wmamills 

Created this PR in openamp-system-reference repo to address this issue: https://github.com/OpenAMP/openamp-system-reference/pull/44

Once that is merged I will create another to remove demos from this repo.

arnopo, 2024-10-25:
Applications are now available in the OpenAMP system reference. This ticket will be closed when the OpenAMP library is cleaned up (target release: 2025.4).

arnopo, 2025-04-17:
Clean-up done in  https://github.com/OpenAMP/open-amp/pull/643

closed 2025-04-17 by arnopo