OpenAMP/open-amp #625: remoteproc/rpmsg: Wrong argument to safe_strcpy()

jonny-svaerd-arm, 2024-10-21:
In commit c3132d0 and 27bec14 the `strncpy` function is replaced by a new `safe_strcpy` function, however the last argument to some of the calls are made with `sizeof(name)` where `name` is a `const char *` (resulting in pointer size).

I'm guessing these most likely were intended to be `strlen(name)`?

Thanks!

arnopo, 2024-10-21:
Good catch!  what i cannot understand  is why the CI does not detect it... 
i will investigate and send a fix

arnopo, 2024-10-21:
@jonny-svaerd-arm: Please,  could you have a look to the fix?

jonny-svaerd-arm, 2024-10-21:
Looks good to me @arnopo. Thanks for fixing it so quickly, much appreciated.

arnopo, 2024-10-21:
> Looks good to me @arnopo. Thanks for fixing it so quickly, much appreciated.

Thank you for reporting the issue! t was good timing as we are starting tests for the upcoming release.

closed 2024-10-25 by arnopo