OpenAMP/open-amp #658: remoteproc_remove_virtio metal_assert(vdev) error

Leo9409, 2025-07-23:
Running openamp-system-reference
```c
void platform_release_rpmsg_vdev(struct rpmsg_device *rpdev, void *platform)
{
	struct rpmsg_virtio_device *rpvdev;
	struct remoteproc *rproc;

	rpvdev = metal_container_of(rpdev, struct rpmsg_virtio_device, rdev);
	rproc = platform;

	rpmsg_deinit_vdev(rpvdev);
	remoteproc_remove_virtio(rproc, rpvdev->vdev);
}
```
rpmsg_deinit_vdev has been rvdev->vdev = NULL; but remoteproc_remove_virtio willbe metal_assert(vdev);

Leo9409, 2025-07-23:
metal: warning:   : freeing non-empty virtqueue
Unable to solve, exception occurred.

tnmysh, 2025-09-07:
@Leo9409 Which release are you using? Could you give top commit hash of openamp-system-reference?

tnmysh, 2025-09-07:
I see the problem now. 

May be virtqueue delete operation should be moved from `rpmsg_deinit_vdev` [1] to `rproc_virtio_remove_vdev` [2]:  Also, vdev shouldn't be assigned NULL in rpmsg_deinit_vdev. rpmsg_deinit_vdev may be can only stop virtio device, but freeing parth remoteproc_virtio should do.

@arnopo, @edmooring do you have any preference ?

[1] https://github.com/OpenAMP/open-amp/blob/98ba93cfabf0a5398e2671b230f94c7cb8b3bef7/lib/remoteproc/remoteproc_virtio.c#L344

[2] https://github.com/OpenAMP/open-amp/blob/98ba93cfabf0a5398e2671b230f94c7cb8b3bef7/lib/remoteproc/remoteproc_virtio.c#L344

arnopo, 2025-09-16:
> I see the problem now.
> 
> May be virtqueue delete operation should be moved from `rpmsg_deinit_vdev` [1] to `rproc_virtio_remove_vdev` [2]: Also, vdev shouldn't be assigned NULL in `rpmsg_deinit_vdev`. rpmsg_deinit_vdev may be can only stop virtio device, but freeing parth remoteproc_virtio should do.
> 
> [@arnopo](https://github.com/arnopo), [@edmooring](https://github.com/edmooring) do you have any preference ?
> 
> We can call, virtio_device_reset during `rpmsg_deinit_vdev`.
> 
> [1]
> 
> [open-amp/lib/rpmsg/rpmsg_virtio.c](https://github.com/OpenAMP/open-amp/blob/98ba93cfabf0a5398e2671b230f94c7cb8b3bef7/lib/rpmsg/rpmsg_virtio.c#L977)
> 
> Line 977 in [98ba93c](/OpenAMP/open-amp/commit/98ba93cfabf0a5398e2671b230f94c7cb8b3bef7)
> 
>  virtio_delete_virtqueues(rvdev->vdev); 
> [2]
> 
> [open-amp/lib/remoteproc/remoteproc_virtio.c](https://github.com/OpenAMP/open-amp/blob/98ba93cfabf0a5398e2671b230f94c7cb8b3bef7/lib/remoteproc/remoteproc_virtio.c#L344)
> 
> Line 344 in [98ba93c](/OpenAMP/open-amp/commit/98ba93cfabf0a5398e2671b230f94c7cb8b3bef7)
> 
>  void rproc_virtio_remove_vdev(struct virtio_device *vdev)

This issue seems already solved in openamp-system-reference using a local variable for vdev in `platform_release_rpmsg_vdev`: https://github.com/OpenAMP/openamp-system-reference/blob/main/examples/legacy_apps/machine/zynqmp/platform_info.c#L284